mobizt/FirebaseJson #2: Int values getting mangled

twisted-techno, 2020-03-27:
Thank you for this library, I have found it useful. 
I have, however found a bug.
Whenever you add integer values that end in more than one zero, the zeroes get removed from that value.  (A value of 1000 becomes a value of 1.)
I can get this to stop happening by removing the call to "_trimDouble(buf);" in the "_addInt" function.
I'm not sure why _trimDouble() is called on an integer string, but it surely mangles certain int values.
I haven't gone further into _trimDouble(), but I bet that it is trying to remove trailing zeroes after a decimal point. 
Float values are added correctly.
I have attached a quick test file to demonstrate the bug.

[ESP32-test.ino.txt](https://github.com/mobizt/FirebaseJson/files/4396212/ESP32-test.ino.txt)

mobizt, 2020-03-28:
Thanks for informing this bug.
I will fix this and inform you about the update.

mobizt, 2020-03-28:
Now I remove the trimDouble from the addInt and setInt functions. 
Please update.

closed 2020-03-29 by mobizt