mobizt/FirebaseJson #4: JSON to URL String Crash

nokemono42, 2020-08-24:
I am trying to interact with an API that uses JSON as a URL variable. The initial API calls are easy to do as URL strings, but the deeper I get into the API the more I have need of converting a FirebaseJson variable into a string. But wjem I do that my NodeMCU crashes. I get a "Soft WDT reset." Initially I thought I was doing something wrong with the watchdog timer, but I have narrowed it down the a memory issue with converting a FirebaseJson object into a string for a URL.

Here is a working URL string. Eventually this just get ugly to work with.

```
String ecobeeHost = "https://api.ecobee.com/";

String url = ecobeeHost + "1/thermostatSummary?json={\"selection\":{\"selectionType\":\"registered\",\"selectionMatch\":\"\"}}";

if (https.begin(client, url)) {
    https.addHeader("Content-Type", "application/json;charset=UTF-8");
    https.addHeader("Authorization", ecobeeBearerData.stringValue + " " + ecobeeCodeData.stringValue);

    int httpsResponseCode = https.GET();
}
```

If I try to make the URL variable a FirebaseJson object and then convert it to a string, it crashes the system.

```
String ecobeeHost = "https://api.ecobee.com/";

FirebaseJson urlJson;
urlJson.set("selection/selectionType", "registered");
urlJson.set("selection/selectionMatch", "");

String urlJsonStr;
urlJson.toString(urlJsonStr, false);

String url = ecobeeHost + "1/thermostatSummary?json=" + urlJson.stringValue;

if (https.begin(client, url)) {
    https.addHeader("Content-Type", "application/json;charset=UTF-8");
    https.addHeader("Authorization", ecobeeBearerData.stringValue + " " + ecobeeCodeData.stringValue);

    int httpsResponseCode = https.GET();
}
```

I have printed the string to the serial console and everything looks fine. I can even run the URL string in an API request tool and get a proper response. Also, I tried to analyze the exception using the exception decoder, but it's never the exact same error. It references the https.GET(), but moving up the stack the error fluctuates.

Is this a bug or am I doing something wrong?

mobizt, 2020-08-24:
There are no bugs in the library.

Due to a lot of changes of HTTPClient and WiFi Client core library, you need to use it correctly.

Please try following code.

```C++
    String ecobeeHost = "https://api.ecobee.com/";

    FirebaseJson urlJson;
    urlJson.set("selection/selectionType", "registered");
    urlJson.set("selection/selectionMatch", "");

    String urlJsonStr;
    urlJson.toString(urlJsonStr, false);

    String url = ecobeeHost + "1/thermostatSummary?json=" + urlJsonStr;

    Serial.println(url);

    //Declare and init the bearSSL SSL client object
    std::unique_ptr<BearSSL::WiFiClientSecure> client(new BearSSL::WiFiClientSecure);

    //Declare the http helper object
    HTTPClient https;

    //If you don't provide the fingerprint to verify or root CA certificate for handshaking
    client->setInsecure();

    //For bearSSL SSL client library, you may need to set the receive and transmit buffers size to limit the memory
    //used by the SSL client to gain more free heap which the default value was 16 kbytes.
    client->setBufferSizes(2048, 2048);

    Serial.println("Begin HTTPS request...");

    if (https.begin(*client, url))
    { 
        https.addHeader("Content-Type", "application/json;charset=UTF-8");
        
        https.addHeader("Authorization", ecobeeBearerData.stringValue + " " + ecobeeCodeData.stringValue);

        int httpsResponseCode = https.GET();

        if (httpsResponseCode > 0)
        {
            Serial.printf("HTTPS GET... code: %d\n", httpsResponseCode);

            if (httpsResponseCode == HTTP_CODE_OK || httpsResponseCode == HTTP_CODE_MOVED_PERMANENTLY)
            {
                //Need to read all server responses from rx buffer before close connection by https.end
                String payload = https.getString();
                Serial.println(payload);
            }
        }
        else
        {
            Serial.println("Error, " + https.errorToString(httpsResponseCode));
        }

        //If no more use of https object, you need to end the https object to release memory.
        https.end();

    }
    else
    {
        Serial.println("Error, unable to connect to server");
    }

```

mobizt, 2020-08-24:
Your both use case are not different and not depend on how you create the url string.

The possible causes of error are memory leak in the defended object that the memory was not released and  if you put your code to the loop, you are  running out of memory.

closed 2020-08-24 by mobizt

nokemono42, 2020-08-26:
Thanks for the quick response. I still couldn't get it to work without memory issues, so I moved to an ESP32. That solved my problem. HTTP json payloads over SSL on a ESP8266 just seem to be too much for it.

Maybe it can work if you really tweak the memory usage, but for developing an application where you don't control the given HTTP response it's really hard to do.

mobizt, 2020-08-26:
The above code was tested and it's ok.

mobizt, 2020-08-26:
You should take care of memory allocation/deallocation and debug the free heap and the stack no matter what the mcu used and memory capacity.

nokemono42, 2020-08-26:
@mobizt Yes, I should. But I'm still prototyping how my application will function. I'm not ready to commit a lot of time to perfecting every HTTP call if it still might change.

mobizt, 2020-08-26:
May be you don't understand what you are using. I mean the WiFi and HTTP Client library which you can manage the request and response your own. 

You can use only WiFiClientSecure class instead of the helper class HTTPClient and WiFiClientSecure to control the request and response header and payload.