mobizt/FirebaseJson #6: I can't compile code for the Teensy 4.0

frmurgia, 2021-02-21:
/libraries/FirebaseJson-2.3.9/src/FirebaseJson.cpp: In member function 'FirebaseJsonArray& FirebaseJsonArray::setJsonArrayData(const String&)':
src/FirebaseJson.cpp:3314:26: error: 'class String' has no member named 'clear'
         data.stringValue.clear();
                          ^

mobizt, 2021-02-21:
Your Arduino installation is not complete.
The data.stringValue is the Arduino String class object as normal use in the Arduino framework.
The clear function is the member function of the Arduino String class.

closed 2021-02-21 by mobizt

mobizt, 2021-02-21:
You need to post the full error here instead of conclude the error because there are many causes of error included third party lib file or class name ambiguous and functions conflict.

frmurgia, 2021-02-21:
thank you for your reply. 

this is the complete message error

```
WARNING: library FirebaseJson-2.3.9 claims to run on esp8266, esp32 architecture(s) and may be incompatible with your current board which runs on avr architecture(s).
/Users/danielemurgia/Documents/Arduino/libraries/libraries/FirebaseJson-2.3.9/src/FirebaseJson.cpp: In member function 'FirebaseJsonArray& FirebaseJsonArray::setJsonArrayData(const String&)':
/Users/danielemurgia/Documents/Arduino/libraries/libraries/FirebaseJson-2.3.9/src/FirebaseJson.cpp:3314:26: error: 'class String' has no member named 'clear'
         data.stringValue.clear();
                          ^
Error compiling for board Teensy 4.0.
```
i Run the basic exemple whit teensy 4.0 board and teensyduino 1.53

mobizt, 2021-02-21:
Due to the WString.h in Teensyduino copied from the AVR core Arduino WString.h and missing the clear function.

You can edit this [line](https://github.com/mobizt/FirebaseJson/blob/master/src/FirebaseJson.cpp#L3314) from

**data.stringValue.clear();**
to
**data.stringValue = "";**