mobizt/FirebaseJson #7: Please check FirebaseJsonHelper::newS

UltimatPronin, 2021-04-01:
Firs of all - thank you for your work and great library! 

Each 7 seconds collect some data & send it to the FireBase.

Data collected this way:
FirebaseJson json;
...
json.set(F("Program"), (int)random(2)); //trouble line (by Decoding stack)
...
then &json passed as parameter to other function.
where I run
Firebase.updateNodeSilent(*gl_fbaseData, MyPath, json);
json.clear();

after random time I get reboot.
I do something wrong or there is problem in FirebaseJsonHelper::newS?

Crashed at 4287096 ms
Restart reason: 2
Exception cause: 29
epc1=0x4000e1b2 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

Decoding stack results
0x402071b5: FirebaseJsonHelper::newS(unsigned int) at D:\Profile\MyUserProfile\Documents\Arduino\libraries\Firebase_ESP8266_Client\src\json/FirebaseJson.h line 358
0x40207ffa: FirebaseJson::_setElementType() at D:\Profile\MyUserProfile\Documents\Arduino\libraries\Firebase_ESP8266_Client\src\json\FirebaseJson.cpp line 2608
0x40207f8d: FirebaseJson::_resetParseResult() at D:\Profile\MyUserProfile\Documents\Arduino\libraries\Firebase_ESP8266_Client\src\json\FirebaseJson.cpp line 2598
0x40208d6a: FirebaseJson::_fbjs_parse(bool) at D:\Profile\MyUserProfile\Documents\Arduino\libraries\Firebase_ESP8266_Client\src\json\FirebaseJson.cpp line 624
0x40242cfa: std::string::_Rep::_M_dispose(std::allocator  const&) at /workdir/arena.x86_64/gcc/xtensa-lx106-elf/libstdc++-v3/include/bits/basic_string.h line 254
0x4020a1e6: FirebaseJson::_set(char const*, char const*) at c:\users\MyUserProfile\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-4-b40a506\xtensa-lx106-elf\include\c++\4.8.2\bits/basic_string.h line 277
0x40242a17: std::string::_S_copy_chars(char*, char const*, char const*) at /workdir/arena.x86_64/gcc/xtensa-lx106-elf/libstdc++-v3/include/bits/basic_string.h line 406
0x402071b5: FirebaseJsonHelper::newS(unsigned int) at D:\Profile\MyUserProfile\Documents\Arduino\libraries\Firebase_ESP8266_Client\src\json/FirebaseJson.h line 358
0x4020a51f: FirebaseJson::_setInt(std::string const&, int) at D:\Profile\MyUserProfile\Documents\Arduino\libraries\Firebase_ESP8266_Client\src\json\FirebaseJson.cpp line 2359
0x402072b4: FirebaseJsonHelper::clearLastError() at D:\Profile\MyUserProfile\Documents\Arduino\libraries\Firebase_ESP8266_Client\src\json/FirebaseJson.h line 284
0x4020a578: FirebaseJson::set(String const&, int) at D:\Profile\MyUserProfile\Documents\Arduino\libraries\Firebase_ESP8266_Client\src\json\FirebaseJson.cpp line 2279
0x4022c568: String::~String() at C:\Users\MyUserProfile\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266\WString.cpp line 125
0x40204cc5: loop() at D:\GIT_Repositories\DSC_Module_ESP8266/DSC_Module_ESP8266.ino line 196
0x4020b833: std::_Function_base::_Base_manager    (FB_RTDB*)> >::_M_manager(std::_Any_data&, std::_Any_data const&, std::_Manager_operation) at c:\users\MyUserProfile\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-4-b40a506\xtensa-lx106-elf\include\c++\4.8.2/functional line 1954
0x4020fb00: FB_RTDB::readStream(FirebaseData*) at D:\Profile\MyUserProfile\Documents\Arduino\libraries\Firebase_ESP8266_Client\src\rtdb\FB_RTDB.cpp line 2189
0x40100b7d: esp_schedule() at C:\Users\MyUserProfile\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266\core_esp8266_main.cpp line 125
0x4022d924: loop_wrapper() at C:\Users\MyUserProfile\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266\core_esp8266_main.cpp line 197

Please check it.

mobizt, 2021-04-01:
The argument type e.g. flash string you provide to the set function is not support (invalid).

The string like types that support by the library and  its functions are Arduino's String object, string literal, pointer to constant char and char array.

Please read the doc [here](https://github.com/mobizt/FirebaseJson/#set-null-to-firebasejson-object-at-the-specified-node-path).

closed 2021-04-01 by mobizt

UltimatPronin, 2021-04-13:
@mobizt , hello again! I changed my code, but still have the error & reboot (time to time).
usage:
FirebaseJson json;
json.set("Ready", (int)random(2));
json.set("Armed", (int)random(2));
json.set("Memory", (int)random(2));
json.set("Bypass", (int)random(2));
json.set("Trouble", (int)random(2)); //trouble line (by Decoding stack)
json.set("Program", (int)random(2));
json.set("Fire", (int)random(2)); 

After random time I get reboot.

Crashed at 18692242 ms
Restart reason: 2
Exception cause: 29
epc1=0x4000e1b2 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

Decoding stack results:
0x4020701d: FirebaseJsonHelper::newS(unsigned int) at D:\Profile\MyUserProfile\Documents\Arduino\libraries\Firebase_ESP8266_Client\src\json/FirebaseJson.h line 358
0x40207e62: FirebaseJson::_setElementType() at D:\Profile\MyUserProfile\Documents\Arduino\libraries\Firebase_ESP8266_Client\src\json\FirebaseJson.cpp line 2609
0x40207df5: FirebaseJson::_resetParseResult() at D:\Profile\MyUserProfile\Documents\Arduino\libraries\Firebase_ESP8266_Client\src\json\FirebaseJson.cpp line 2599
0x40208bd2: FirebaseJson::_fbjs_parse(bool) at D:\Profile\MyUserProfile\Documents\Arduino\libraries\Firebase_ESP8266_Client\src\json\FirebaseJson.cpp line 625
0x402426ea: std::string::_Rep::_M_dispose(std::allocator  const&) at /workdir/arena.x86_64/gcc/xtensa-lx106-elf/libstdc++-v3/include/bits/basic_string.h line 254
0x4020a056: FirebaseJson::_set(char const*, char const*) at c:\users\MyUserProfile\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-4-b40a506\xtensa-lx106-elf\include\c++\4.8.2\bits/basic_string.h line 277
0x40242407: std::string::_S_copy_chars(char*, char const*, char const*) at /workdir/arena.x86_64/gcc/xtensa-lx106-elf/libstdc++-v3/include/bits/basic_string.h line 406
0x4020701d: FirebaseJsonHelper::newS(unsigned int) at D:\Profile\MyUserProfile\Documents\Arduino\libraries\Firebase_ESP8266_Client\src\json/FirebaseJson.h line 358
0x4020a3df: FirebaseJson::_setInt(std::string const&, int) at D:\Profile\MyUserProfile\Documents\Arduino\libraries\Firebase_ESP8266_Client\src\json\FirebaseJson.cpp line 2360
0x4020711c: FirebaseJsonHelper::clearLastError() at D:\Profile\MyUserProfile\Documents\Arduino\libraries\Firebase_ESP8266_Client\src\json/FirebaseJson.h line 284
0x4020a438: FirebaseJson::set(String const&, int) at D:\Profile\MyUserProfile\Documents\Arduino\libraries\Firebase_ESP8266_Client\src\json\FirebaseJson.cpp line 2280
0x4022c24a: String::String(char const*) at C:\Users\MyUserProfile\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266\WString.cpp line 36
0x40204bb5: loop() at D:\GIT_Repositories\DSC_Module_ESP8266/DSC_Module_ESP8266.ino line 195
0x4020b6f3: std::_Function_base::_Base_manager    (FB_RTDB*)> >::_M_manager(std::_Any_data&, std::_Any_data const&, std::_Manager_operation) at c:\users\MyUserProfile\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-4-b40a506\xtensa-lx106-elf\include\c++\4.8.2/functional line 1954
0x4020f900: FB_RTDB::handleStreamRead(FirebaseData*) at D:\Profile\MyUserProfile\Documents\Arduino\libraries\Firebase_ESP8266_Client\src\rtdb\FB_RTDB.cpp line 2198
0x40100b7d: esp_schedule() at C:\Users\MyUserProfile\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266\core_esp8266_main.cpp line 125
0x4022d37c: loop_wrapper() at C:\Users\MyUserProfile\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266\core_esp8266_main.cpp line 197

Please take a look again.

mobizt, 2021-04-13:
Your device is running out of free heap or stack overflow (in case these are in the callback function).

Check the free heap or try to move the code to some function that calls by the main loop task.

mobizt, 2021-04-13:
The esp8266 stack size is only 4k and you should avoid doing something that required large memory in any callback functions.

The callback functions included the Firebase-ESP8266 stream callback functions and your own defined callback functions and any third party libraries callback functions which asynchronous called.

UltimatPronin, 2021-04-13:
1) Code fragment is in the main loop
2) I have average this results: Free heap: 7912 bytes (fragmentation: 02 %).  Is it not enought?
3) I will check callback functions too

mobizt, 2021-04-13:
The error is memory allocation failed which may grow the stack or reduce the free heap.

7192 is relatively low memory and it's critical for internal system tasks included WiFi which would be failed to perform.

You should keep your free heap at least 10k.

UltimatPronin, 2021-04-13:
Thank you! I will try enlarge free heap first. Or may be is better to move to esp32?

mobizt, 2021-04-13:
I can't tell because it's your code.