mobizt/FirebaseJson #10: Push data stream to Firebase using ArduinoJSON Library Object

HarborDesigns, 2021-12-07:
Hello,
I am storing nested JSON array on my SD card. I am using [ArduinoJSON library](https://github.com/bblanchon/ArduinoJson) to serialize and deserialize my data. I am using JSON streaming method to deserialize the data from SD card. I have verified that all the code related to ArduinoJSON library works. I am not sure how to transfer the ArduinoJSON Stream object to FirebaseJSON object. Could you show me a quick example to do it. I will write down my pseudo code which will help you understand what I am trying to do.

**Pseudo Code**
```
file = SD.open("/testfilename.json",FILE_READ);
ReadBufferingStream bufferedReadFile{file, 64};
DynamicJsonDocument doc(24576);
deserializeJson(doc, bufferedReadFile);

FirebaseJson json;
json.set(bufferedReadFile);
Serial.printf("Push file... %s\n",  Firebase.RTDB.pushJSON(&fbdo, "/test/array", &json); ? "ok" : fbdo.errorReason().c_str());

bufferedReadFile.flush();
file.close();
```

**Questions:**

1. Do I need to deserialize the SD card JSON file or can I simply Stream it to Firebase. Is there any faster way to do it. If I create another buffer then I am afraid there might be memory leak
2. I am reading lots of data collected over hours from SD card and hence I have to parse the data and upload it in chunks. Could you recommend me a a way to send the data in a loop continuously. I can read the data from SD card and print it in my serial terminal in chunks but I am not sure if I can do the same using Firebase library.  
3. Can you show me example on how to stream data to my Firebase server. I currently get compile error where I am trying to push data to the server

 
**Here is the environment that I'm using:**
Microconroller: ESP32
Core/runtime: ESP32core for Arduino v2.0.1
IDE: Arduino IDE 1.8.13
ArduinoJSON library 6.18.4
FirebaseJSON: 2.6.1
Firebase ESP32 Client: 3.12.6

mobizt, 2021-12-07:
FirebaseJson accepted string for deseialize or parsing e.g. Flash string, Arduino String, std::string, char array and string literal, by using setJsonData method or pass to the constructor.

https://github.com/mobizt/FirebaseJson/blob/64f0b975f65215fd4b0280510ee57c3112e1fbc4/examples/BasicUsage/Create/Create.ino#L27-L34

It also supports data from [client and  stream (File and Serial)](https://github.com/mobizt/FirebaseJson/tree/master/examples).

In [File.ino](https://github.com/mobizt/FirebaseJson/blob/master/examples/Stream/File/File.ino) example, it reads the stream a  single char, one by one to determine the end of valid JSON , then parse from internal buffered data.

Other objects or classes from other libraries are not supported due to unknown interfaces.

HarborDesigns, 2021-12-07:
I commented my ArduinoJSON deserialization and deserialized it using Firebase. I can print the content of my file to serial terminal but I don't know how to push the data to the Firebase server.

```
FirebaseJson json;
json.readFrom(file);
json.toString(Serial,true); //this will print to serial terminal
Serial.println(); 
Serial.printf("Push file... %s\n",  Firebase.RTDB.pushString(&fbdo, "/test/array", &json ); ? "ok" : fbdo.errorReason().c_str());
```

mobizt, 2021-12-07:
json is object not a string.

https://github.com/mobizt/Firebase-ESP32/tree/master/src#append-new-child-node-key-and-value-using-firebasejson-object-to-the-defined-database-path

closed 2021-12-07 by mobizt