mobizt/FirebaseJson #11: error: 'class FIREBASE_CLASS' has no member named 'Firestore'

abdullah230695, 2022-02-07:
**FirebaseNew:206:22: error: 'class FIREBASE_CLASS' has no member named 'Firestore'
         if (Firebase.Firestore.createDocument(&fbdo, FIREBASE_PROJECT_ID, "" /* databaseId can be (default) or empty */, documentPath.c_str(), content.raw()))**

I can push data to RTDB but can't in Firestore. 

**compile error : error: 'class FIREBASE_CLASS' has no member named 'Firestore'**

pls check my code below

/**
 * Created by K. Suwatchai (Mobizt)
 * 
 * Email: k_suwatchai@hotmail.com
 * 
 * Github: https://github.com/mobizt/FirebaseJson
 * 
 * Copyright (c) 2022 mobizt
 *
*/

#if defined(ESP32)
#include <WiFi.h>
#include <FirebaseESP32.h>
#elif defined(ESP8266)
#include <ESP8266WiFi.h>
#include <FirebaseESP8266.h>
#endif

//Provide the token generation process info.
#include <addons/TokenHelper.h>

//Provide the RTDB payload printing info and other helper functions.
#include <addons/RTDBHelper.h>

#include <Firebase_ESP_Client.h>

/* 1. Define the WiFi credentials */
#define WIFI_SSID "WIFI_AP"
#define WIFI_PASSWORD "WIFI_PASSWORD"

//For the following credentials, see examples/Authentications/SignInAsUser/EmailPassword/EmailPassword.ino

/* 2. Define the API Key */
#define API_KEY "web_api_key"

/* 3. Define the RTDB URL */
#define DATABASE_URL "db_url" //<databaseName>.firebaseio.com or <databaseName>.<region>.firebasedatabase.app

/* 4. Define the user Email and password that alreadey registerd or added in your project */
#define USER_EMAIL "mail"
#define USER_PASSWORD "pass"

/* 5. Define the project ID */
#define FIREBASE_PROJECT_ID "id"

//Define Firebase Data object
FirebaseData fbdo;

FirebaseAuth auth;
FirebaseConfig config;

unsigned long sendDataPrevMillis = 0;

unsigned long count = 0;

unsigned long dataMillis = 0;

void setup()
{

    Serial.begin(115200);

    WiFi.begin(WIFI_SSID, WIFI_PASSWORD);
    Serial.print("Connecting to Wi-Fi");
    while (WiFi.status() != WL_CONNECTED)
    {
        Serial.print(".");
        delay(300);
    }
    Serial.println();
    Serial.print("Connected with IP: ");
    Serial.println(WiFi.localIP());
    Serial.println();

    Serial.printf("Firebase Client v%s\n\n", FIREBASE_CLIENT_VERSION);

    /* Assign the api key (required) */
    config.api_key = API_KEY;

    /* Assign the user sign in credentials */
    auth.user.email = USER_EMAIL;
    auth.user.password = USER_PASSWORD;

    /* Assign the RTDB URL (required) */
      config.database_url = DATABASE_URL;

    //Or use legacy authenticate method
      config.signer.tokens.legacy_token = "my_token";

    Firebase.begin(&config, &auth);

    //Comment or pass false value when WiFi reconnection will control by your code or third party library
    Firebase.reconnectWiFi(true);
    
    //For sending payload callback
    //config.cfs.upload_callback = fcsUploadCallback;
}

void loop()
{
    //Flash string (PROGMEM and  (FPSTR), String C/C++ string, const char, char array, string literal are supported
    //in all Firebase and FirebaseJson functions, unless F() macro is not supported.

    if (Firebase.ready() && (millis() - sendDataPrevMillis > 15000 || sendDataPrevMillis == 0))
    {
        sendDataPrevMillis = millis();

        FirebaseJson json;
        json.setDoubleDigits(3);
        json.add("temp", count);
        json.add("humidity", count+1);
        json.add("CO2", count+2); //generating random number
        json.add("Tvoc", count+3);
        json.add("pm25", count+4);
        json.add("pm10", count+5);
        json.add("aqi", count+6);

        Serial.printf("Set json... %s\n", Firebase.RTDB.setJSON(&fbdo, "/Devices/DeviceId_1/", &json) ? "ok" : fbdo.errorReason().c_str());

        Serial.printf("Get json... %s\n", Firebase.RTDB.getJSON(&fbdo, "/test/json") ? fbdo.to<FirebaseJson>().raw() : fbdo.errorReason().c_str());

       /* FirebaseJson jVal;
        Serial.printf("Get json ref... %s\n", Firebase.RTDB.getJSON(&fbdo, "/test/json", &jVal) ? jVal.raw() : fbdo.errorReason().c_str());

        FirebaseJsonArray arr;
        arr.setFloatDigits(2);
        arr.setDoubleDigits(4);
        arr.add("a", "b", "c", true, 45, (float)6.1432, 123.45692789);

        Serial.printf("Set array... %s\n", Firebase.RTDB.setArray(&fbdo, "/test/array", &arr) ? "ok" : fbdo.errorReason().c_str());

        Serial.printf("Get array... %s\n", Firebase.RTDB.getArray(&fbdo, "/test/array") ? fbdo.to<FirebaseJsonArray>().raw() : fbdo.errorReason().c_str());

        Serial.printf("Push json... %s\n", Firebase.RTDB.pushJSON(&fbdo, "/test/push", &json) ? "ok" : fbdo.errorReason().c_str());

        json.set("value", count + 0.29745);
        */
       // Serial.printf("Update json... %s\n\n", Firebase.RTDB.updateNode(&fbdo, "/test/push/" + fbdo.pushName(), &json) ? "ok" : fbdo.errorReason().c_str());

        count++;
    }

    
    if (Firebase.ready() && (millis() - dataMillis > 60000 || dataMillis == 0))
    {
        dataMillis = millis();

        //For the usage of FirebaseJson, see examples/FirebaseJson/BasicUsage/Create.ino
        FirebaseJson content;

        //We will create the nested document in the parent path "a0/b0/c0
        //a0 is the collection id, b0 is the document id in collection a0 and c0 is the collection id in the document b0.
        //and d? is the document id in the document collection id c0 which we will create.
        String documentPath = "a0/b0/c0/d" + String(count);

        //If the document path contains space e.g. "a b c/d e f"
        //It should encode the space as %20 then the path will be "a%20b%20c/d%20e%20f"

        //double
        content.set("fields/myDouble/doubleValue", 123.45678);

        //boolean
        content.set("fields/myBool/booleanValue", true);

        //integer
        content.set("fields/myInteger/integerValue", "911");

        //null
        content.set("fields/myNull/nullValue"); // no value set

        String doc_path = "projects/";
        doc_path += FIREBASE_PROJECT_ID;
        doc_path += "/databases/(default)/documents/coll_id/doc_id"; //coll_id and doc_id are your collection id and document id

        //reference
        content.set("fields/myRef/referenceValue", doc_path.c_str());

        //timestamp
        content.set("fields/myTimestamp/timestampValue", "2014-10-02T15:01:23Z"); //RFC3339 UTC "Zulu" format

        //bytes
        content.set("fields/myBytes/bytesValue", "aGVsbG8="); //base64 encoded

        //array
        content.set("fields/myArray/arrayValue/values/[0]/stringValue", "test");
        content.set("fields/myArray/arrayValue/values/[1]/integerValue", "20");
        content.set("fields/myArray/arrayValue/values/[2]/booleanValue", true);

        //map
        content.set("fields/myMap/mapValue/fields/name/stringValue", "wrench");
        content.set("fields/myMap/mapValue/fields/mass/stringValue", "1.3kg");
        content.set("fields/myMap/mapValue/fields/count/integerValue", "3");

        //lat long
        content.set("fields/myLatLng/geoPointValue/latitude", 1.486284);
        content.set("fields/myLatLng/geoPointValue/longitude", 23.678198);

        count++;

        Serial.print("Create a document... ");

        if (Firebase.Firestore.createDocument(&fbdo, FIREBASE_PROJECT_ID, "" /* databaseId can be (default) or empty */, documentPath.c_str(), content.raw()))
            Serial.printf("ok\n%s\n\n", fbdo.payload().c_str());
        else
            Serial.println(fbdo.errorReason());
    }
    }

mobizt, 2022-02-07:
Wrong place to open the issue.

Make sure you install this library https://github.com/mobizt/Firebase-ESP-Client

mobizt, 2022-02-07:
You can't use Firestore with the libraries FirebaseESP32 and FirebaseESP8266.
These libraries are for RTDB only.

closed 2022-02-07 by mobizt

abdullah230695, 2022-02-07:
sorry..