mobizt/FirebaseJson #13: Add support for strings from flash

sh-user, 2022-02-08:
Hi is it possible to add strings from flash for get and set function 
For example `json.get(result, PSTR("temp"));` or `json.get(result, FPSTR("temp"));`

sh-user, 2022-02-08:
Works `json.get(res, String(PSTR("temp")).c_str());` but probably there is another solution.

mobizt, 2022-02-08:
It's already supported.

Which device you used which you think it's not supported.

sh-user, 2022-02-08:
I have an ESP8266, if I write a PSR it causes a reboot.

sh-user, 2022-02-08:
nonos sdk 2.2.1 (legacy) 
Arduino IDE 1.8.13

mobizt, 2022-02-08:
For ESP8266 core version2.7.x and earlier I will add support flash string later.

mobizt, 2022-02-08:
It is currently supported FPSTR which F and PSTR are not supported.

sh-user, 2022-02-08:
Works like this
```
//global variables
const char F_temp[] PROGMEM = "temp";
uint8_t tmp = 0;
...
json.get(result, String(FPSTR(F_temp)).c_str());
tmp = result.to<int>();
```

mobizt, 2022-02-08:
More simple with 
**json.get(result, pgm2Str(F_temp));**

pgm2Str is the macro defined in the library.

mobizt, 2022-02-08:
The PGM_P (PROGMEM) **will not supported** as function arguments by the library due to it is ambiguous with the function that accepted const char*.

For F and PSTR macros will be supported later.

closed 2022-02-08 by mobizt