mobizt/FirebaseJson #18: Read from Google sheet

diegostf, 2022-04-17:
Testing the example to read a cell from spreadsheets, I get the following result:

```c++
 bool success = GSheet.values.append(&response /* returned response */, "1z_YTLhUXAXXPMpPSCK7SjS0DoMaWB50pt7X4HqSHgXs" /* spreadsheet Id to append */, "Log1!A7" /* range to append */, &valueRange /* data range to append */);
      response.toString(Serial, true);
      Serial.println();

      Serial.println("\nGet spreadsheet values from range...");
      Serial.println("---------------------------------------------------------------");

      success = GSheet.values.get(&response /* returned response */, "1z_YTLhUXAXXPMpPSCK7SjS0DoMaWB50pt7X4HqSHgXs" /* spreadsheet Id to read */, "Log1!B3" /* range to read */);
      response.toString(Serial, true);
      Serial.println("\n---------------------");
      
      FirebaseJsonData result;

      response.get(result, "values");
      if(result.success)
      {
         //Print its content e.g.string, int, double, bool whereas object, array and null also can access as string
         Serial.println(result.to<String>());     
      }else Serial.println("No result"); 
 ```

The result is this:

[image]

How can I get an integer (in this case "25")?

mobizt, 2022-04-17:
Use 

`[response.get(result, "values/[0]/[0]")`

And cast the result to int as

`result.to <int>()`

mobizt, 2022-04-17:
For FirebaseJson usage, see the examples in examples folder in this repository.

diegostf, 2022-04-17:
> Use
> 
> `[response.get(result, "values/[0]/[0]")`
> 
> And cast the result to int as
> 
> `result.to <int>()`

I tried that, but the result was the same ( 0 ).

```c++
FirebaseJsonData result;

      response.get(result, "values/[0]/[0]");
      if(result.success)
      {
         //Print its content e.g.string, int, double, bool whereas object, array and null also can access as string
         Serial.println(result.to<int>());
     
      }else Serial.println("No result");  
 ```

[image]

I also tried the examples you mentioned, but still no success.

I think the problem is the spreadsheet's response. The formatting is as follows: [["25"]]. These square brackets are causing the problem.

mobizt, 2022-04-18:
Nothing wrong with the FirebaseJson as the type of data is string then convert to integer will be 0 and you need to cast it using String class like this.

`Serial.println(result.to<String>().toInt())`

closed 2022-04-18 by mobizt

diegostf, 2022-04-18:
> Nothing's wrong with the FirebaseJson as the type of data is string then convert to integer will be 0 and you need to cast it using String class like this.
> 
> `Serial.println(result.to<String>().toInt())`

That's great. Now it works smooth
Thanks a lot