mobizt/FirebaseJson #24: bug: exception on ESP32-WROVER with PSRAM enabled

ilka-schulz, 2023-07-14:
This library in version `3.0.7` works fine on ESP32-WROOM but it crashes (thread panic) on [ESP32-WROVER](https://www.espressif.com/sites/default/files/documentation/esp32-wrover-e_esp32-wrover-ie_datasheet_en.pdf) (an ESP32 variant with 8 MB PSRAM).

#### MWE Crashing Code

```c++
#include <FirebaseJson.h>

void setup()  {
    FirebaseJson json;
    json.set("foo","bar");  // <-- This line throws an exception.
}
void loop() {
}
```

To use the PSRAM on ESP32-WROVER, I added all these compiler flags:
- `-DBOARD_HAS_PSRAM`
- `-mfix-esp32-psram-cache-issue`
- `-DFIREBASEJSON_USE_PSRAM`

#### What I tried
- removing `-DFIREBASEJSON_USE_PSRAM`: does not fix the problem
- removing all of the above compiler flags: fixes the problem but I can not use PSRAM anymore...

#### Further Information

- FirebaseJson version: `3.0.7`
- board: [Olimex ESP32-POE-ISO-WROVER](https://www.olimex.com/Products/IoT/ESP32/ESP32-POE-ISO/open-source-hardware)

mobizt, 2023-07-14:
FIREBASEJSON_USE_PSRAM is not compiler flag to be use like that.

It is the macro you should define in FBJS_Config.h.

closed 2023-07-14 by mobizt

ilka-schulz, 2023-07-16:
I think there is a misunderstanding:

1. I checked the `FBJS_Config.h` and it already contains:
```c++
/** Use PSRAM for supported ESP32/ESP8266 module */
#if defined(ESP32) || defined(ESP8266)
#define FIREBASEJSON_USE_PSRAM
#endif
```
According to my IDE diagnostics, `ESP32` is defined and `FIREBASEJSON_USE_PSRAM` is defined as well.

2. The `-D...` compiler flag adds a preprocessor define to all compiled code so it should do the same.

### Workaround

I found that removing `FIREBASEJSON_USE_PSRAM` both from `FBJS_Config.h` and from the compiler flag serves as a workaround. This required me to remove the above `#if defined (ESP32).....` lines. I can now use the PSRAM for my application and FB will not use it.

This really looks like it is a bug in FB, not a configuration error on my side, right?

mobizt, 2023-07-17:
Ther is no misunderstanding. You said right about macro and compiler flag, but you should define `FIREBASEJSON_USE_PSRAM` in the library config file only instead of using it as compiler flag to prevent unknown error as there are some other macros defined in the core which this library depends on.

mobizt, 2023-07-17:
I also test the library with ESP32-WROVER (8MB PSRAM) module, and it works fine with default config on `FBJS_Config.h` and not need to modify anything.

The `MB_String` class is actually used to handle dynamic memory allocation in this `FirebaseJson` library and in all my PSRAM capable libraries without issue.

In your case I'm not sure why your device crashed, it could be device defects.

You should add some level debug in compiler flag to see the result of `psramInit()` like this.

`-DCORE_DEBUG_LEVEL=5`

Note that, the memory allocation for external SPI ram in ESP32 Arduino framework is only support maximum 4MB using `ps_malloc` function used in this `FirebaseJson` library.

mobizt, 2023-07-17:
Test this simple code without changing the library config with ESP32 core v2.0.9 in ESP32-WROVER (8MB PSRAM)

```cpp
#include <Arduino.h>
#include <FirebaseJson.h>

void setup()
{
    Serial.begin(115200);
}

void loop()
{
    int ps_free_size = ESP.getFreePsram();
    int heap_free_size = ESP.getFreeHeap();
    
    FirebaseJson json;
    json.add(F("key"), F("__________________________________________________"));

    printf("PSRAM used %u\n", ps_free_size - ESP.getFreePsram());
    printf("Heap used %u\n", heap_free_size - ESP.getFreeHeap());
    Serial.println();

    delay(2000);
}

```

And this is result.

```yaml
[    24][D][esp32-hal-cpu.c:244] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
[   468][I][esp32-hal-psram.c:96] psramInit(): PSRAM enabled
PSRAM used 212
Heap used 100

PSRAM used 212
Heap used 0

PSRAM used 212
Heap used 0

PSRAM used 212
Heap used 0

PSRAM used 212
Heap used 0

PSRAM used 212
Heap used 0

PSRAM used 212
Heap used 0

PSRAM used 212
Heap used 0

PSRAM used 212
Heap used 0

PSRAM used 212
Heap used 0

```

mobizt, 2023-07-17:
> I can now use the PSRAM for my application and FB will not use it.

Without your clear and simple code and test result, I'm not sure and cannot prove that PSRAM works in your application.

If you don't define the macro `FIREBASEJSON_USE_PSRAM`, the library will use internal heap instead using normal C/C++ `malloc` which you can test with my example code that posted above.

mobizt, 2023-07-17:
Library does not intend to allow user to use macro defined in config file as a compiler flag which it can lead to unknown error as it does not guard correctly, and you should follow the library documentation and examples for correct usage.