mobizt/FirebaseJson #26: Got an error while trying to use FirebaseJson library along the FirebaseClient library

capc0307, 2024-06-19:
Hi,

I got the following error when trying to upload my code into my ESP32-S3 with the Arduino IDE. I'm trying to use the [FirebaseJson](https://github.com/mobizt/FirebaseJson) library along the new [FirebaseClient](https://github.com/mobizt/FirebaseClient/tree/main) library. How can I solve this problem? I've used FirebaseJson library along the now deprecated [Firebase-ESP-Client](https://github.com/mobizt/Firebase-ESP-Client) library in the past and it worked perfectly.

Thanks in advance,

capc0307.

capc0307, 2024-06-19:
```
In file included from d:\Users\usuario\Documents\Arduino\libraries\FirebaseJson\src/FirebaseJson.h:135,
                 from D:\Users\usuario\Downloads\FirebaseClient\FirestoreDatabase\get\get.ino:43:
d:\Users\usuario\Documents\Arduino\libraries\FirebaseJson\src/MB_String.h:115:22: error: expected nested-name-specifier before 'enable_if'
  115 | #define MB_ENABLE_IF enable_if
      |                      ^~~~~~~~~
d:\Users\usuario\Documents\Arduino\libraries\FirebaseJson\src/MB_String.h:115:22: note: in definition of macro 'MB_ENABLE_IF'
  115 | #define MB_ENABLE_IF enable_if
      |                      ^~~~~~~~~
d:\Users\usuario\Documents\Arduino\libraries\FirebaseJson\src/MB_String.h:115:22: error: expected initializer before 'enable_if'
  115 | #define MB_ENABLE_IF enable_if
      |                      ^~~~~~~~~
d:\Users\usuario\Documents\Arduino\libraries\FirebaseJson\src/MB_String.h:115:22: note: in definition of macro 'MB_ENABLE_IF'
  115 | #define MB_ENABLE_IF enable_if
      |                      ^~~~~~~~~
d:\Users\usuario\Documents\Arduino\libraries\FirebaseJson\src/MB_String.h:115:22: error: expected nested-name-specifier before 'enable_if'
  115 | #define MB_ENABLE_IF enable_if
      |                      ^~~~~~~~~
d:\Users\usuario\Documents\Arduino\libraries\FirebaseJson\src/MB_String.h:115:22: note: in definition of macro 'MB_ENABLE_IF'
  115 | #define MB_ENABLE_IF enable_if
      |                      ^~~~~~~~~
d:\Users\usuario\Documents\Arduino\libraries\FirebaseJson\src/MB_String.h:115:22: error: expected initializer before 'enable_if'
  115 | #define MB_ENABLE_IF enable_if
      |                      ^~~~~~~~~
d:\Users\usuario\Documents\Arduino\libraries\FirebaseJson\src/MB_String.h:115:22: note: in definition of macro 'MB_ENABLE_IF'
  115 | #define MB_ENABLE_IF enable_if
      |                      ^~~~~~~~~
d:\Users\usuario\Documents\Arduino\libraries\FirebaseJson\src/MB_String.h:115:22: error: expected nested-name-specifier before 'enable_if'
  115 | #define MB_ENABLE_IF enable_if
      |                      ^~~~~~~~~
d:\Users\usuario\Documents\Arduino\libraries\FirebaseJson\src/MB_String.h:115:22: note: in definition of macro 'MB_ENABLE_IF'
  115 | #define MB_ENABLE_IF enable_if
      |                      ^~~~~~~~~
d:\Users\usuario\Documents\Arduino\libraries\FirebaseJson\src/MB_String.h:115:22: error: expected initializer before 'enable_if'
  115 | #define MB_ENABLE_IF enable_if
      |                      ^~~~~~~~~
d:\Users\usuario\Documents\Arduino\libraries\FirebaseJson\src/MB_String.h:115:22: note: in definition of macro 'MB_ENABLE_IF'
  115 | #define MB_ENABLE_IF enable_if
      |                      ^~~~~~~~~
d:\Users\usuario\Documents\Arduino\libraries\FirebaseJson\src/MB_String.h: In constructor 'MB_String::MB_String(bool)':
d:\Users\usuario\Documents\Arduino\libraries\FirebaseJson\src/MB_String.h:549:18: error: call of overloaded 'appendNum(bool&)' is ambiguous
  549 |         appendNum(value);
      |         ~~~~~~~~~^~~~~~~
d:\Users\usuario\Documents\Arduino\libraries\FirebaseJson\src/MB_String.h:927:16: note: candidate: 'MB_String& MB_String::appendNum(float, int)'
  927 |     MB_String &appendNum(float value, int precision = 5)
      |                ^~~~~~~~~
d:\Users\usuario\Documents\Arduino\libraries\FirebaseJson\src/MB_String.h:941:16: note: candidate: 'MB_String& MB_String::appendNum(double, int)'
  941 |     MB_String &appendNum(double value, int precision = 9)
      |                ^~~~~~~~~
d:\Users\usuario\Documents\Arduino\libraries\FirebaseJson\src/MB_String.h:955:16: note: candidate: 'MB_String& MB_String::appendNum(long double, int)'
  955 |     MB_String &appendNum(long double value, int precision = 9)
      |                ^~~~~~~~~
d:\Users\usuario\Documents\Arduino\libraries\FirebaseJson\src/MB_String.h: In member function 'MB_String& MB_String::operator+=(bool)':
d:\Users\usuario\Documents\Arduino\libraries\FirebaseJson\src/MB_String.h:777:18: error: call of overloaded 'appendNum(bool&)' is ambiguous
  777 |         appendNum(value);
      |         ~~~~~~~~~^~~~~~~
d:\Users\usuario\Documents\Arduino\libraries\FirebaseJson\src/MB_String.h:927:16: note: candidate: 'MB_String& MB_String::appendNum(float, int)'
  927 |     MB_String &appendNum(float value, int precision = 5)
      |                ^~~~~~~~~
d:\Users\usuario\Documents\Arduino\libraries\FirebaseJson\src/MB_String.h:941:16: note: candidate: 'MB_String& MB_String::appendNum(double, int)'
  941 |     MB_String &appendNum(double value, int precision = 9)
      |                ^~~~~~~~~
d:\Users\usuario\Documents\Arduino\libraries\FirebaseJson\src/MB_String.h:955:16: note: candidate: 'MB_String& MB_String::appendNum(long double, int)'
  955 |     MB_String &appendNum(long double value, int precision = 9)
      |                ^~~~~~~~~
d:\Users\usuario\Documents\Arduino\libraries\FirebaseJson\src/MB_String.h: In member function 'MB_String& MB_String::appendPtr(mb_string::MB_StringPtr)':
d:\Users\usuario\Documents\Arduino\libraries\FirebaseJson\src/MB_String.h:869:22: error: call of overloaded 'appendNum(long long unsigned int&)' is ambiguous
  869 |             appendNum(*addrTo<uint64_t *>(src.address()));
      |             ~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
d:\Users\usuario\Documents\Arduino\libraries\FirebaseJson\src/MB_String.h:927:16: note: candidate: 'MB_String& MB_String::appendNum(float, int)'
[... 758 lines cut ...]
[... cut ...]
d:\Users\usuario\Documents\Arduino\libraries\FirebaseJson\src/FirebaseJson.h: In instantiation of 'bool FirebaseJson::get(FirebaseJsonData&, T, bool) [with T = const char*]':
D:\Users\usuario\Downloads\FirebaseClient\FirestoreDatabase\get\get.ino:204:38:   required from here
d:\Users\usuario\Documents\Arduino\libraries\FirebaseJson\src/FirebaseJson.h:2469:46: error: 'getStr' was not declared in this scope
 2469 |         bool ret = mGet(root, &result, getStr(path, addr), prettify);
      |                                        ~~~~~~^~~~~~~~~~~~

exit status 1

Compilation error: 'class FirebaseJsonData' has no member named 'to'
```

capc0307, 2024-06-19:
The code I'm trying to implement is a modification of the example from the [FirebaseClient](https://github.com/mobizt/FirebaseClient/tree/main) library made by myself. This is the [original example](https://github.com/mobizt/FirebaseClient/blob/main/examples/FirestoreDatabase/Documents/Async/NoCallback/Get/Get.ino).

I got the error as I included the FirebaseJson library, added additional values to the  Cloud Firestore database and modified the printResult function like this:

```
void printResult(AsyncResult &aResult)
{
    if (aResult.isEvent())
    {
        Firebase.printf("Event task: %s, msg: %s, code: %d\n", aResult.uid().c_str(), aResult.appEvent().message().c_str(), aResult.appEvent().code());
    }

    if (aResult.isDebug())
    {
        Firebase.printf("Debug task: %s, msg: %s\n", aResult.uid().c_str(), aResult.debug().c_str());
    }

    if (aResult.isError())
    {
        Firebase.printf("Error task: %s, msg: %s, code: %d\n", aResult.uid().c_str(), aResult.error().message().c_str(), aResult.error().code());
    }

    if (aResult.available())
    {
        Firebase.printf("task: %s, payload: %s\n", aResult.uid().c_str(), aResult.c_str());

        FirebaseJson firestoreDocumentDataJSON;
        FirebaseJsonData result;
        String name = "", lastName= "";

        firestoreDocumentDataJSON.setJsonData(aResult.c_str());
        firestoreDocumentDataJSON.get(result, "fields/name/stringValue");

        if(result.success){
          name= result.to<String>();
          Serial.print("The name is: ");
          Serial.println(name);
        }

        else{
          Serial.println("Error 1.");
        }

        firestoreDocumentDataJSON.get(result, "fields/lastName/stringValue");

        if(result.success){
          lastName= result.to<String>();
          Serial.print("The last name is: ");
          Serial.println(lastName);
        }

        else{
          Serial.println("Error 2.");
        }
        
    }
}
```

mobizt, 2024-06-19:
It is not support and cannot use,

closed 2024-06-19 by mobizt

capc0307, 2024-06-19:
> You can update the `FirebaseJson` library to [v3.0.9](https://github.com/mobizt/FirebaseJson/releases/tag/v3.0.9) to fix the issue.
> 
> It will be available from Library Manager soon.

Hi, mobitz.

I just updated the library to v3.0.9, tested it and it works perfectly. Thank you so much for your help.