mobizt/FirebaseJson #27: Async Push Callback Called Twice/Wrong: Heap Corrupt After Log Upload (double-free)

peterzu1, 2025-05-14:
On ESP32 with FirebaseClient (version 2.0.3), I'm using async log upload via database.push(..., onPushComplete).  
After the log queue is empty and I print "All log entries uploaded", the callback is invoked a second time (even though I never call push again).

In the second callback call, I have a guard that ensures my logic is not run twice. Still, shortly after, the heap is corrupted as shown below.

This looks like internal memory is being freed twice. If I remove log upload or don't use the async callback, this does not occur.

Here's my guard and the printed output:

void onPushComplete(AsyncResult &res) {
    if (logUploadComplete) {
        Serial.println("[GUARD] log upload already completed, skipping callback.");
        return;
    }
    ...
    if (logQueue.empty()) {
        ... // cleanup
        logUploadComplete = true;
        firebaseOpState = OP_IDLE;
    }
}  

[printed output cut for brevity, but includes:]
✅ All log entries uploaded.
🧹 Local log cleared after Firebase upload.
[GUARD] log upload already completed, skipping callback. 
CORRUPT HEAP: Bad head at 0x3ffd56b8. Expected 0xabba1234 got 0x3ffca42c
...

mobizt, 2025-05-14:
No, that is your code issue not library issue.

The callback is just the function to get the result when the process is completed.

closed 2025-05-14 by mobizt

mobizt, 2025-05-14:
The error is because out of memory or memory leak that causes memory allocation failure, then something does not allocate due to failure, then try to de-allocate, this type of multiple heap de-allocation error can be occurred.

You have to debug your code.